EVNotify/EVNotify #224: E-Golf Integration

tpreugsc, 2021-01-29:
### New issue
- Short description: Integration des E-Golf in die Liste verfügbarer Fahrzeuge
- Detailed description:

- Can you reproduce it? (_If so, how?_):

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  
Wie kann ich helfen, den E-Golf in die Liste zu bekommen. Ich kann leidlich Quellcode lesen und kämpfe mich durch die ID3 vue.

Weiterhin kann ich google ganz gut utzen (:-) ) und benötigte Befehle raussuchen.

Ein Beispiel:
ATZ
ATE0
AT TP 6
ATH1
AT CRA 7ED
AT SH 7E5
22028C55555555
AT AR

und bekomme ein Byte Nutzdaten zurück:
7ed 04 62 02 8c b9

Das rechne ich dann wie folgt um:
(Wert - 16) / 2,24

Das ergibt den Ladezustand in %.

Was wird jedoch benötigt, um möglichst die anderen Werte noch zu bekommen? Primär brauche ich nur SoC für ABRP, der Rest interessiert mich eigentlich nicht/bekomme ich auch woanders her. ABer wenn, dann richtig, gell?

</details>

tpreugsc, 2021-01-29:
Kann es sein, das Ihr im Moment beim ID3 auch nur den SOC auslest?

GPlay97, 2021-01-29:
I'll write in English since it should always be in English here. 

Yes, soc currently only for id3. I'm working on more..

Which e golf do you have? A working version has already been verified (2017+ version) with one build here (but not with your commands). Maybe your stuff is for older e golfs?

tpreugsc, 2021-01-29:
I have e-golf 300, made in 2020. where can i find this version you mentioned? as stated above, just the soc would be enough. but it should be the soc from the carnet module, as it is readable with ignition off. problem is, its the soc brutto (35 kwh) not netto (31,5kwh) of the battery, but math can probably help out.

mem-, 2021-04-01:
> I'll write in English since it should always be in English here.
> 
> Yes, soc currently only for id3. I'm working on more..
> 
> Which e golf do you have? A working version has already been verified (2017+ version) with one build here (but not with your commands). Maybe your stuff is for older e golfs?

I also have an e-Golf (2020) and wonder how do I find the build and how to install it on my Android phone?

tpreugsc, 2021-04-01:
Just using the id3 profile. Set battery capacity to 31 kwh. Getting soc and battery temp. Enough for me. Charging power would be nice, but I will probably survive not having that information....

mem-, 2021-04-06:
Thanks tpreugsc, I'll try that later today.

Raudi1, 2021-04-07:
There will be a proper e-golf profile soon according to the developer. So just keep your eyes open for an update of the app.

mem-, 2021-04-08:
@tpreugsc I tested the ID.3 profile and with my car parked I got similar remaining distance as the car dashboard when I entered 28 kWh. Temperatures were something like 2/1/-342 C :-)
I guess I can tweak the battery size after some trips and charging cycles.

@Raudi1 looking forward to the e-Golf profile.

Raudi1, 2021-04-08:
The range is just a simple calculation of entered battery size and consumption afaik.Will learn from past drives in the future I hope.

tpreugsc, 2021-04-19:
Some news: soc is readable via obd while charging. Remember to code out the obd interface from the alarm system surveillance. Otherwise you get alarms when charging with AC. Didnt had that problem with DC charging.

ari197, 2021-08-05:
Hi .. which OBD dongle do you use to connect ? 

have you tried to integrate this with ABRP ?

tpreugsc, 2021-08-05:
> 
> 
> Hi .. which OBD dongle do you use to connect ?
> 
> have you tried to integrate this with ABRP ?

I'm using an adapter called viecar 4.0. This one works with android and ios.

And yes, I have it integrated with abrp. Shows the SOC and ABRP reroutes based on SOC as it see fit. Sadly, charging speed can't be calculated as it isn't read from BMS correctly. You have to disable the live reference consumption data send from EVnotify to abrp in abrp. See my settings in ABRP below.
[image]
[image]

ari197, 2021-08-05:
why do we have to disable the live consumption from evNotify ?

tpreugsc, 2021-08-05:
you DID looked at the first screenshot, right? Because there are no usable values! 4458 Wh/km? Really? That's why we wait for a e-Golf profile in EV Notify....

ari197, 2021-08-05:
ah oke .. understood ..