EVNotify/EVNotify #230: Mustang Mach-e integration

sebastienroyal, 2021-03-05:
### New issue
- Short description: when do you expect to add functionnality for the 2021 Ford Mach-e?
- Detailed description:

- Can you reproduce it? (_If so, how?_):

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

lcamata, 2021-04-26:
I'm interested in this topic as well, and I'm also offering to collect OBD data from the car if you need any.

toskium, 2021-09-19:
I am interested as well. If I can assist with providing data, please let me know.

martiinezz, 2021-11-26:
@lcamata @toskium do you have good OBD2 USB dongle and latest Forscan?

toskium, 2021-11-27:
@martiinezz I do have an ODBLink MX. Is this the tool you suggest: https://play.google.com/store/apps/details?id=com.astech.forscanlite if so, no issue I'll fetch it.

toskium, 2021-11-27:
@martiinezz I do have an OBDLink EX and an MX for wireless use with my phone. I could grab the latest version of Forscan as well. However I am not a OBD professional, so you'd need to point me into the right direction.

martiinezz, 2021-11-28:
@toskium you can try Forscan Lite with your ODBLink to see if your car is recodnized - later if possible I would go windows laptop + forscan + usb kable like ELS-37

toskium, 2021-11-29:
@martiinezz Forscan Lite seems to work, the info view shows my car and modules:

```
Adapter: OBDLink MXBTr2.2 STN1151 v4.5.0 (ELM327 v1.3a)
Min.delay: 7 ms (Excellent)
Errors: None

Model: Ford Mach-E
Engine: Battery Electric Vehicle
Capacity: 75-99kWh
Year: 2021
Generation: 2021 MY
Fuel: Gasoline
Transmission gear: Automatic
...
```

martiinezz, 2021-12-12:
Now you would have to capture some CAN traffic - you would need some extra hardware connected to laptop to do that.

martiinezz, 2021-12-25:
Just recently ABRP added ODB2 support for their app - and it supports Mach-E from day 1. You should try it to see if it works for you.

toskium, 2022-01-19:
@martiinezz thanks, I needed to order a new OBD dongle since my current one wasn't compatible. With the OBDlink CX everything works as expected, you have to update to the latest firmware trough their app, though.

Maybe as a tip for other MME drivers: you need to add a new vehicle to ABRP, you can't link the dongle to an existing vehicle. Took me a while... 

The issue can be closed now.