EVNotify/EVNotify #231: Volkswagen e-up! support

romnGit, 2021-03-06:
Hello,

How can we help add support for the e-up! 2.0 (and 1.0 ?) 

Here is some documentation : 
https://docs.openvehicles.com/en/latest/components/vehicle_vweup/docs/index.html
https://docs.openvehicles.com/en/latest/components/vehicle_vweup/docs/index_obd.html
https://www.goingelectric.de/wiki/Liste-der-OBD2-Codes/
https://github.com/sharkcow/VW-e-UP-OBD-CAN-logs

HabibA007, 2021-04-14:
If I can support this topic, please ask. I‘m not a developer only an e-up driver in very close future. But I have an seat mii electric for testing which is an identical car only different brand

Raudi1, 2021-06-13:
The VW owned brand cars are all closely related. The ID3 profile mostly works for the e-golf. You can also try it for your cars.

markus-k, 2021-08-31:
I just tested, the ID.3 profile doesn't work with the e-up (2020). SoC is shown as NaN and all other numbers are 0. Not sure why, because the commands used should be the same, with the exception of `0322189d55555555` (`22 18 9d`) not being mentioned in the GoingElectric wiki.

markus-k, 2021-09-02:
The problem was that the e-up only seems to support 11-bit CAN while ID.3 and eGolf use 29-bit CAN (I'm not an expert on this), the the init sequence and commands are slightly different. I have created a pull request with basic support for e-up: #265