EVNotify/EVNotify #234: bundle.js does not get built

pegr69, 2021-04-02:
### bundle.JS does not get built

If I check out your code and build according to instructons the bundle.js file does not get generated and the screen is blank, all files load except that one.
I have never used the vue frameworks so am a bit lost here.

Can you give me a hint?

PS
Thx for a great app! Works fine with the ID3, except for some small issues, which I was planning to help out with.
DS

SamuelBrucksch, 2021-04-23:
When I tried to build, it asked me to add a Google-Services.json and also the client.js had wrong formatting with the colons. How did you get that far to build an app?

GPlay97, 2021-04-23:
You need Rollup (something like webpack) to generate this file. Do cd app. And there rollup -c