EVNotify/EVNotify #237: Add VW ID.4 Experimental support

MattiJarvinen, 2021-04-30:
### New issue
- Short description: 

ID4 experimental support could be added based on ID3 template. 

- Detailed description:

ID.Furkan tested basic information on ID.4 with ID.3 template and it seemed to work [EVNotify Works! | VW ID.4 1st Max](https://www.youtube.com/watch?v=wSZzirCYQ1Y&t=47s)
It seemed to be working too with Tesla Björn's [VW ID4 1st 82 kWh range test](https://www.youtube.com/watch?v=QOwDEjelgFE&t=76s)

Maybe it will work on any VW using the same platform?

ID4 battery sizes are according to EV-database.org: 

|Name| Usable | Total |
|-------|---------|-------|
| [1st](https://ev-database.org/car/1273/Volkswagen-ID4-1st) | 77kWh | 82kWh |
| [Pro Performance](https://ev-database.org/car/1314/Volkswagen-ID4-Pro-Performance) | 77kWh | 82kWh  |
| [Pure](https://ev-database.org/car/1489/Volkswagen-ID4-Pure) | 52kWh | 55kWh  |
| [Pure Performance](https://ev-database.org/car/1411/Volkswagen-ID4-Pure-Performance) | 52kWh | 55kWh  |

\* **For very critical security breaches, please mail directly at info@evnotify.de**

StorageFan, 2021-06-05:
I second that request and would like to extend it to also adding the Skoda ENYAQ as a selectable car to the app well. 
The ID.4 and ENYAQ are pretty much brothers and share the same MEB-platform with the ID.3. What makes it even easier is that that the battery specs are completely identical to the ID.4, although my sources (including EV-database.org, ADAC and elektroauto-news.net) provide slightly different numbers for the batteries than mentioned in the table above:

Name | Usable (net) | Full (gross)
-- | -- | --
ENYAQ iV 50 | 52 kWh | 55 kWh
ENYAQ iV 60 | 58 kWh | 62 kWh
ENYAQ iV 80 | 77 kWh | 82 kWh
ENYAQ iV Sportline 60 | 58 kWh | 62 kWh
ENYAQ iV Sportline 80 | 77 kWh | 82 kWh

GPlay97, 2021-08-01:
Added in #258

closed 2021-08-01 by GPlay97