EVNotify/EVNotify #240: Crash on iOS with BT OBD2 Dongle

RalphS69, 2021-05-18:
### New issue
- critical issue 
- follow-up on https://github.com/EVNotify/EVNotify/issues/215
- during initialization the app either crashes or runs into an initialization error

#### Hardware / Software:
- iPhone 6s with iOS 14.5.1
- OBD2 dongle: vLinker MC+ for iOS and Android
- Ioniq elctric classic
<details>
<summary>no successful connection</summary>

- The dongle is listed in the settings as **vLinker MC-IOS**
- when going back to the dashboard, the app
  - crashes, or
  - repeatedly says:
    - "initialization, please wait"
    - "OBDII error. Re-initializing, Please wait"
</details>

GPlay97, 2021-05-18:
Is car fully turned on or charging?

Otherwise the dongle may not be supported, on iOS only one dongle is confirmed to work yet: https://github.com/EVNotify/EVNotify/wiki/Compatibility

RalphS69, 2021-05-18:
- If the car is fully turned on: The app crashes
- If the car is off, the initialization fails repeatedly

I updated the initial comment accordingly.

Why is just one device supported? My dongle was not a cheap one, and works fine with car scanner.

GPlay97, 2021-05-18:
Crash is weird, but I don't know if I have the chance to reproduce it locally. Will see in the upcoming days when I find time. On iOS, only one ble characterises can be used currently, technical limitation. This can only change in the upcoming v3 Version with native code

RalphS69, 2021-05-18:
Yeah. Occasionally I see quite short the "initialization, please wait" - than this app is gone. Would be nice if you could improve this somehow - I need this app to feed abrp with the actual car data.
BTW: Is the app still available for iOS? I cannot find it in the store anymore ...

GPlay97, 2021-05-18:
Currently not, was taken away a few days ago by Apple due to expired credit card. It's expected that it'll be available again until end of this week hopefully

RalphS69, 2021-06-27:
> I've finally got the right dongle and get data/log with IOS from an
> ioniq 28kwh and also see the app crashing after a few minutes. I could
> not figure out what actually triggers this. Any way to debug this?

Not from my side. I am just the user who would like to use the app ...

oscjan, 2022-11-25:
Same problem for me on iPhone 13 running ios 16.1.1 and OBD2 dongle: vLinker MC+ for iOS and Android. Cars VW ID.4 GTX.

m3gg3, 2023-02-28:
same problem here. ios 16.3.1 and a supported dongle https://www.amazon.de/gp/product/B071D8SYXN/ref=as_li_ss_tl?ie=UTF8&psc=1&linkCode=sl1&tag=gplay97-21&linkId=68ed6e726985a720808ee22317d51f8b&language=de_DE

app opens then initialize and crashes

RalphS69, 2023-07-13:
evNotify is not available in the AppStore anymore (at least in DE). Why?

The app could not be used directly (due to the crash), but I could use it as remote monitor, retrieving the data in the car with an Android phone.

So if just this issue is the reason of the removal: Please add it again, its useful even with the BT connection not working.