EVNotify/EVNotify #242: Read cell volt and temp

flopp999, 2021-05-26:
Anyone know how to get those data through OBD.
It seems to be possible, at least with ID3.

GPlay97, 2021-05-26:
You have a reference? It's already supported for some cars.

flopp999, 2021-05-26:
I found a link about Nissan Leaf

https://www.mynissanleaf.com/viewtopic.php?t=11676

flopp999, 2021-05-26:
> You have a reference? It's already supported for some cars. 

https://www.facebook.com/ID.3LifeSweden/photos/a.111367054443898/116962510551019/?type=3

spot2000, 2021-05-26:
I have all the data for cell volt and temp for ID.3 because I supplied it all to EVNotify.

It will be publicly accessible for all on my GitHub soon.

Mvh
Conny (I am ID.3 life Sweden 😉)

Skickat från min iPhone 11 Pro Max

https://www.facebook.com/ID.3LifeSweden/photos/a.111367054443898/116962510551019/?type=3