EVNotify/EVNotify #243: Tutorial for new vehicles

jcerdan, 2021-05-29:
Could you create a tutorial to add new vehicles? or does it exist somewhere and I haven't seen it?

I've got a Skoda Citigo e iV and it would be great to connect it with EVNotify/ABRP as this car has very little information showed for users. 
I tried to set ID3 as it's the same group (VW)  but data is not the same and nothing was displayed.
I can try to collect the needed info.

Regards

GPlay97, 2021-05-29:
I can do that, especially in the upcoming new app generation. 

Basically the PIDs are needed to retrieve state of charge and so on. This needs to exist first, before support can be added

jcerdan, 2021-05-29:
ok, 
I'll try to get this information.
thanks

jcerdan, 2021-05-30:
I have found this information:
https://www.goingelectric.de/wiki/Liste-der-OBD2-Codes/#Liste-der-OBD2-Codes-VW-e-Up
https://github.com/sharkcow/VW-e-UP-OBD-CAN-logs/blob/master/carscanner.csv
Is that the info you need?
I've got OBD eleven if you need me to check for something.

regards,