EVNotify/EVNotify #245: Does OBDLink MX+ work on iOS?

mrubioroy, 2021-05-30:
OBDLink MX+ is certified “Made for iPhone”. Probably the only Bluetooth (not LE) working on iOS.
Does it work with EVNotify on iOS?
Is EVNotify for iOS only available on TestFlight?
What are the advantages of using Bluetooth vs Bluetooth LE?

mrubioroy, 2021-07-15:
I've tested and it doesn't.

I see there's a Cordova plugin that would allow the connection to "Made for iPhone" classic Bluetooth accessories on iOS:
https://github.com/stevef51/cordova-ios-external-accessories

The API is similar to BluetoothSerial.

GPlay97, 2022-07-30:
The new app generation, v3, that is under active development, as mentioned now in the README, supports a lot (if not all) dongles , even under iOS. v2 only works with limited dongles unfortunately.

closed 2022-07-30 by GPlay97