EVNotify/EVNotify #248: Dutch translation

Bishamon1987, 2021-06-20:
Hey,
I'm working on a NL.js
The dutch translation for the app. 
I don't know how to make branches so there a way for me to send this file? 
Or shall I paste the code in here?

Bishamon1987, 2021-06-21:
Hereby I send you the Dutch Translation.
If strings need to be added, let me know :)

`export default {
    UPDATE_AVAILABLE_TITLE: 'Nieuwe update',
    UPDATE_AVAILABLE_TEXT: 'Er is een nieuwe update beschikbaar. Update de app alstublieft en profiteer van de nieuwste functies en oplossingen.',
    TOUR_LOGIN_1: 'Welkom bij EVNotify. Met EVNotify kunt u op afstand de gegevens van uw elektrische voertuig controleren!',
    TOUR_LOGIN_2: 'Om direct te beginnen kunt u hier een account aanmaken.',
    TOUR_LOGIN_3: 'Als u al gebruik maakt van EVNotify, u kunt uw bestaande inloggegevens gebruiken. Daarna kunt u de gegevens van de auto volgen!',
    TOUR_SETTING_1: 'U heeft nu de mogelijkheid om instellingen te maken. U kunt deze op elk moment aanpassen. Hier bijvoorbeeld uw taal.',
    TOUR_SETTING_2: 'Om EVNotify te laten werken, moet u een auto selecteren. Controleer of u de juiste auto heeft geselecteerd. Als uw auto niet in de lijst voorkomt, is deze helaas nog niet compatibel.',
    TOUR_SETTING_3: 'Hier kunt u uw gemiddeld verbruik aangeven. Dit kan op elk moment opnieuw worden aangepast en dient voor de berekening van het bereik.',
    TOUR_SETTING_4: 'EVNotify heeft een OBD2 Bluetooth-dongle nodig om te werken. Deze moet eerder zijn gekoppeld in de systeeminstellingen. Kies deze. Staat het niet in de lijst, klik dan op onderstaande knop. Mogelijk moet u de app opnieuw starten als de lijst na het koppelen niet wordt bijgewerkt.',
    TOUR_SETTING_5: 'Stel hier de waarde in wanneer u een melding wilt ontvangen. Onthoud dat u deze waarde op elk moment kunt aanpassen. De types kunnen later worden gewijzigd.',
    TOUR_SETTINGS_1: 'Zoals eerder vermeld, kunnen alle instellingen altijd worden gewijzigd. Vergeet niet om hier op het pictogram Opslaan te klikken na het wijzigen.',
    TOUR_SETTINGS_2: 'Als er onbehandelde fouten optreden die mogelijk niet worden opgemerkt, wordt informatie doorgegeven aan EVNotify voor een betere probleemoplossing. Dit kan hier worden uitgeschakeld.',
    TOUR_SETTINGS_3: 'Hier kunt u uw account beheren. Ook kunt u hier uw AKey vinden om op andere apparaten in te loggen.',
    TOUR_SETTINGS_4: 'Om ritten als een logboek te laten verschijnen, moet dit worden geactiveerd. Hiermee wordt de locatie overgedragen. Dit is een lokale instelling waarvoor toegang tot de locatie vereist is.',
    TOUR_SETTINGS_5: 'Als dit geactiveerd is, wordt Bluetooth automatisch geactiveerd wanneer Bluetooth uitgeschakeld is.',
    TOUR_SETTINGS_6: 'Als EVNotify na een paar minuten stopt met monitoren, kan deze optie worden ingeschakeld. Hierdoor blijft het scherm actief en worden de energiebesparende instellingen van het apparaat omzeild.',
    TOUR_SETTINGS_7: 'Indien actief, start EVNotify automatisch wanneer het apparaat opnieuw wordt opgestart.',
    TOUR_SETTINGS_8: 'Het huidige versienummer. Het gerucht gaat dat hier iets gebeurt, als u er een beetje op klikt.',
    TOUR_DASHBOARD_1: 'De pictogrammen tonen de huidige verbindingsstatus. Als de wolk naar boven wijst, worden gegevens verzonden, anders worden ze ontvangen. Als een van de symbolen is doorgestreept, is er geen netwerk-/Bluetooth-/serververbinding. Het keversymbool wordt gebruikt voor het oplossen van problemen.',
    TOUR_DASHBOARD_2: "De actuele gegevens, die door de auto worden uitgelezen of ontvangen, worden hier getoond. U kunt door de verschillende pagina's scrollen om meer gegevens te zien.",
    TOUR_DASHBOARD_3: 'Via de balk hieronder vindt de navigatie plaats. Kijk eens rond! Bedankt voor het gebruik van EVNotify!',
    V1_MIGRATION_NOTE: 'Als u voorheen versie 1 gebruikte, dan moet u een nieuw account aanmaken.',
    SLOGAN: 'Monitor uw elektrische auto',
    NEXT: 'Volgende',
    CANCEL: 'Annuleren',
    INVALID_CREDENTIALS: 'Ongeldige inloggegevens. Controleer uw ingevoerde inloggegevens.',
    UNEXPECTED_ERROR: 'Onverwachte fout',
    CHECK_INPUTS: 'Controleer inputs',
    CHECK_PASSWORD: 'Wachtwoord moet minimaal 6 karakters lang zijn',
    PASSWORD: 'Wachtwoord',
    PASSWORD_ENTER: 'Vul uw huidige wachtwoord in.',
    PASSWORD_NEW: 'Kies een nieuw wachtwoord van minimaal 6 karakters. Controleer uw wachtwoord voordat u het aanpast!',
    PASSWORD_REPEAT: 'Herhaal wachtwoord',
    PASSWORD_MISMATCH: 'Wachtwoorden komen niet overeen',
    PASSWORD_CHANGE: 'Pas wachtwoord aan',
    PASSWORD_CHANGED: 'Wachtwoord succesvol aangepast.',
    TOKEN_RESET: 'Reset token',
    TOKEN_RESET_WARNING: 'Weet u zeker dat u uw token wil resetten? Hierdoor wordt uw huidige token ongeldig en hebben andere apparaten en applicaties die de oude token gebruiken geen toegang meer totdat u zich opnieuw aanmeldt.',
    TOKEN_RESETTED: 'Token succesvol gereset',
    CREATE_ACCOUNT: 'Maak account aan',
    PASSWORD_FORGOT: 'Wachtwoord vergeten',
    CHANGE_LANGUAGE: 'Pas taal aan',
    LOGIN: 'Inloggen',
    LOGOUT: 'Uitloggen',
    PROCEED: 'Doorgaan',
    DONE: 'Klaar',
    SAVE: 'Opslaan',
    INTRODUCTION: 'Introductie',
    SETUP: 'Instellen',
    OWN_RISK: 'Gebruik op eigen risico. Ik ben niet aansprakelijk voor schade die het gevolg kan zijn van oneigenlijk gebruik of het gebruik van een nagemaakte OBD2-dongle. Door een account aan te maken gaat u hiermee akkoord.',
    INTRODUCTION_TEXT_1: 'Met EVNotify kunt u uw elektrische voertuig op afstand bewaken.',
    INTRODUCTION_TEXT_2: 'Keep track of your state of charge and let you notify when your desired state of charge is reached.',
[... cut ...]
    LOG_DELETED: 'Logbestand invoer verwijderd',
    LOG_DELETED_ERROR: 'Logbestand invoer kan niet worden verwijderd',
    EXPORT: 'Exporteren',
    EXPORT_LOG: 'Exporteer logbestand',
    EXPORT_LOG_TEXT: 'U kunt deze logbestanden exporteren met alle bijbehorende gegevens. Dit zal een bestand genereren dat naar een specifiek e-mailadres wordt verzonden. Dit proces kan enkele minuten inbeslag nemen.',
    LOG_EXPORTED: 'Logbestand geëxporteerd. Bekijk uw e-mail',
    LOG_EXPORTED_ERROR: 'Fout tijdens exporteren van het logbestand'
};
`

iGadget, 2022-09-27:
Too bad this contribution has just been sitting here for well over a year without any response. IMHO that's not the way to motivate contributors... How can we help improve this situation?

GPlay97, 2022-09-29:
Sorry for not responding to this issue to that time.

Lately I'm focusing on completely developing a new app version (server and backend with new technology and tech stack). For now, the repository is private source, but it will be open source again with contributions and ideas welcomed.

I wanted to focus on that first, but only mentioned it within a notice in README here, that v3 currently has high priority. It is planned to test with few people in the next weeks. Currently it is in an internal test phase already.