EVNotify/EVNotify #249: iPhone new version?

flopp999, 2021-06-23:
I could not find any other way to contact you so I created an issue.

Will there be a new version for iPhone with support for Skoda Enyaq or VW ID.3/4?

HabibA007, 2021-07-04:
It seems the project is dead?

GPlay97, 2021-07-04:
> It seems the project is dead?

No. iOS is ready, I'm running the latest version on my iPhone for latest tests. Still not really stable, working on v3, new native Version. I don't know if it will pass apple review, they always find something.. or if new iOS version will come with v3

swifty-tim, 2021-07-10:
Do you need any help it? I am a IOS engineer, let me know if I can help with it

HabibA007, 2021-07-13:
> > It seems the project is dead?
> 
> No. iOS is ready, I'm running the latest version on my iPhone for latest tests. Still not really stable, working on v3, new native Version. I don't know if it will pass apple review, they always find something.. or if new iOS version will come with v3

what is the difference between “nothing happens” and a dead project?

GPlay97, 2021-07-13:
> what is the difference between “nothing happens” and a dead project?

You have a bad day? Or why you post here too besides the other issue?

You can follow progress for the migrations. For your other issue: choose id3 profile to retrieve data or wait until I release the new cars, e.g the ze50 (first code already here, rest with internal testers). Besides the migration and updates, there is also a lot of support , daily mails and so on that I do (at least one hour after work)

HabibA007, 2021-07-14:
I want to apologize for my tone. I'm just a frustrated user who uses an Android as a send phone on his Enyaq and his defaul smartphone is an IOS. Every time you were in the IOS settings, the car switches to "auto" and you no longer get any output on the IOS. Therefore I wait like many others for an "ID3" or better "Enyaq" support of the IOS client. In the meantime, IOS developers have also made support offers here.

Sorry again for my tone and thank you again for your time and work on the project.

tolnaiz, 2021-08-04:
@GPlay97 I can help you testing the iPhone version if you are interested. I have an iPhone 12, an OBDLink MX+ and an ID3. I am able to build it from source, I built a couple of flutter applications.

mrubioroy, 2021-08-17:
Can we participate in a TestFlight beta test? I don't find any branch on GitHub with the latest iOS native code to build myself.

Soulplayer, 2021-09-05:
Hello,

I also want to use TestFlight if this is possible. 

Thank you!

holdinghausen, 2021-09-15:
I'd love to betatest, too. Electronics Engineer, Enyaq, iPhone... :-)

HabibA007, 2021-12-28:
Is it realistic that there will be another IOS version with MEB support?
In the meantime, I can do without my Android phone for evnotify/ABRP, since ABRP offers native OBD BLE adapter support and that also under iOS. 
Of course, it would be nice if I could continue to use EVNotify.  But for that, the IOS version would finally need MEB support.

GPlay97, 2021-12-29:
> Is it realistic that there will be another IOS version with MEB support?
>
> In the meantime, I can do without my Android phone for evnotify/ABRP, since ABRP offers native OBD BLE adapter support and [...]

Thanks for your patience!

I just made it right now. I'll publish a new version on the app store. Hopefully it will passes the review fast. I could also fix the crash issue on iOS now. Very nice. I'm hyped. 

Sorry that you waited Soo long. I'm back!

HabibA007, 2022-01-06:
I've been checking for the update every day since then but so far nothing. Did it not pass the check?

RualdOrdelman, 2022-02-13:
Same for me. Waiting for the new iOS version with the Enyaq fix. @GPlay97

GPlay97, 2022-07-30:
It was released in the meantime. A complete new (and way better) app version (v3) is in development, as mentioned now in the README.

closed 2022-07-30 by GPlay97