EVNotify/EVNotify #253: Data log automation

CarlosAF72, 2021-07-20:
### New issue
- Short description: 
- Detailed description:

- Can you reproduce it? (_If so, how?_):

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

GPlay97, 2021-07-20:
To be able to automatically log drives, you have to enable location synchronization. Ensure it's enabled in EVNotify and within system settings of your phone (requires high precision mode). After drive (at least a few minutes, at least 5), it's shortly created afterwards.

iGadget, 2022-09-27:
I can't seem to get this working. I've enabled local synchronization, but I'm never getting anything in the logs. Not in Charges, not in Drives and not in Statistics. What am I doing wrong?

GPlay97, 2022-09-29:
If requirements are met (a couple of minutes of transferring data), the log should be created automatically after a couple of minutes. For driving logs, it is currently important to enable high precision, so location data can be retrieved.

You can also write a mail with your akey to info(at) evnotify.de, so I can have a look at it. Just then send me a date, where you charged for example.

closed 2022-09-29 by GPlay97