EVNotify/EVNotify #254: Incompatible Android 10 Tablet

EVfrank, 2021-07-22:
### Incompatible Device
- Short description: Android 10 Device Incompatible
- Detailed description: Hyjoy Android 10 Tablets, 8" IPS Full HD Touchscreen, 2GB RAM, 32GB Storage, Quad-Core Processor, Long Battery Life, Dual Camera, Wi-Fi, Bluetooth will not download EVNotify. It cannot find it in the Google Play store. When linking to the Play store through the EVNotify.de link, the app begins to download and then returns the error message: "No eligible devices for app install. This item is not compatible with your device"

- Can you reproduce it? (_If so, how?_):

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):No
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

GPlay97, 2021-07-23:
Does it have Google Android System Webview installed? It requires Google services in order to work

EVfrank, 2021-07-23:
> Does it have Google Android System Webview installed? It requires Google services in order to work

Thanks for the quick reply, yes Webview is installed + enabled (91.0.4472.164). From the Google Play Store, the following messages are returned: "Your device isn't compatible with this version" and "This app may not be optimized for your device."

GPlay97, 2021-07-26:
Can you try to install the Apk file manually? You find it here under releases, just to see if this is a play store thing or a real incompatibility. Thanks in advance