EVNotify/EVNotify #261: Failing to integrate with ABRP

wabiloo, 2021-08-12:
### New issue
- Short description: EVNotify on Android tablet, gets ERR_FILE_NOT_FOUND when trying to integrate with ABRP
- Detailed description:
I have an Android tablet on which ABRP and EVNotify are installed. EVNotify is connected to Vgate BLE dongle. When I go to the settings to connect to ABRP, I get a black screen with a popup saying "net:ERR_FILE_NOT_FOUND (file://android_asset/www[object%20Window])

- Can you reproduce it? (_If so, how?_):
Yes. Kill the app, start again, go to Integrations, click on ABRP, click Next in the first popup

- [x] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):  Yes. Only reason I use EVNotify is to connect it to ABRP
  
 <details>
  <summary>additional information</summary>
  
AKey = c1d453

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

sirdir1972, 2022-01-02:
I have the same problem on an Carlinkit AI Box

philos31, 2022-11-17:
I have the same issue.

Vkaba, 2023-10-05:
i got the same issue.

App Version 2.4
ABRP App Version 4.6.2 (2445)
Android 8 (i know, old phone)

joopdo, 2024-07-08:
Same issue for me!