EVNotify/EVNotify #267: 28kWh, Logging stops after 10min while charging

esni, 2021-09-12:
Is this a car / obd adapter limitation or related to EVnotify?

I'm using an android 10 phone permanently present in the car, connected to wifi.

mhantsch, 2021-09-14:
I found that under Android 10 (In my case LineageOS), EVNotify can get stopped / suspended if it is running in the background. I worked around this using an Automate script which keeps activating EVNotify to keep it in the foreground. See here: https://www.goingelectric.de/wiki/EVNotify-Alternative-Loesung-mit-Automate/

GPlay97, 2022-07-30:
There is also a way to bypass energy saving options with the following "hack":

Go to EVNotify settings, go down to version number. Click on it multiple times fast. You'll enter a new screen. Activate background mode (and optionally persistent notifications too). Save. Go back to dashboard. It should now no longer fall asleep.

closed 2022-07-30 by GPlay97