EVNotify/EVNotify #277: HomeAssistant added to repo?

joopdo, 2022-01-30:
I'm writing a HomeAssistant frontend to the EVNotify API.

Would you be interested to receive the code/readme as a .zip to add it to a additional repo under the EVNotify account? (I can do PR for future releases)
My thinking is that it's tied to EVNotify anyway and will contribute to EVNotify usecases, when it's here hopefully there will be others who will contribute to it as well.

Current state: 
[image]

Todo:
* Clean up friendly names
* Add proper Icons
* Add 'car selection' method to be used for capacity calculations. (Currently I'm not interested in that as I have one car)

d3wy, 2022-02-01:
Watching with keen interest! Please let me know if you might like some assistance with documentation or similar.

anthonws, 2022-02-15:
@joopdo Looks awesome! How are you getting the data?

SebGla, 2023-02-26:
@joopdo did you finally create a repo or custom integration ?