EVNotify/EVNotify #281: Android app white screen after build - missing build.js?

joopdo, 2022-02-08:
Build with following:
```
Ubuntu 20.04.3 LTS
Cordova 11
Gradle 5.6.3
node v10.19.0
npm 6.14.4
```

To build, I followed `CONTRIBUTIONS.md` with some additions:

```
cordova platform add android@8.0.0
cordova plugin rm cordova-plugin-fcm

```

After opening the app, EVNotify opens but screens stays white. logcat shows:

```
02-08 11:41:06.723  9293  9293 D CordovaActivity: Resumed the activity.
02-08 11:41:06.803  1718  1842 I ActivityTaskManager: Displayed com.evnotify.app/.MainActivity: +337ms
02-08 11:41:06.817  9293  9293 D CordovaWebViewImpl: onPageDidNavigate(file:///android_asset/www/index.html)
02-08 11:41:06.837  9293  9322 E AndroidProtocolHandler: Unable to open asset URL: file:///android_asset/www/client.js
02-08 11:41:06.843  9293  9322 E AndroidProtocolHandler: Unable to open asset URL: file:///android_asset/www/js/bundle.js
02-08 11:41:07.109  9293  9293 D JsMessageQueue: Set native->JS mode to EvalBridgeMode
02-08 11:41:07.112  9293  9293 D CordovaWebViewImpl: onPageFinished(file:///android_asset/www/index.html)
02-08 11:41:07.113  9293  9370 E AndroidProtocolHandler: Unable to open asset URL: file:///android_asset/www/cordova_plugins.js
```
@GPlay97  Any pointers here? When I move `client.js` from root to ./www, one error disappears. Seems the app can't reach `bundle.js`. Build log throws some errors but nothing critical on first sight. Tried with latest master branch and EVNotify-2.2.1-patch.3.

ivanfmartinez, 2022-02-09:
which version of android SDK and java are you using ?

I'm trying to compile without success and will try to replicate your environment on a docker container.

GPlay97, 2022-02-09:
Did you run rollup before compiling? Otherwise there will be no source code transpiled

joopdo, 2022-02-09:
@GPlay97 
I didn't know about rollup. Ran `rollup -c --environment INCLUDE_DEPS,BUILD:production` in the `app` folder and now the build works. Yay! Closing issue. On a separate note, I'll e-mail you because I have a question about the interesting product [EVNotify PlugAndPlay](https://shop.evnotify.com/#buy)

@ivanfmartinez 
Java:
`echo $JAVA_HOME             
/usr/lib/jvm/java-17-oracle`

Android SDK:
`Android 8.0 (Oreo) API 26' and 'Android API 32' are installed currently.

closed 2022-02-09 by joopdo

ivanfmartinez, 2022-02-09:
@joopdo can you run the command

sdkmanager  --list_installed

to show all installed android versions in your environment ?

joopdo, 2022-02-10:
@ivanfmartinez Sorry, sdkmanager doesn't work on my machine. The information above is what I got from Android Studio. Here's a screenshot...

[image]