EVNotify/EVNotify #287: Missing Android release tag/APK

IzzySoft, 2022-05-01:
On PlayStore, you've released v2.3.0 a while ago. Could you please add the corresponding app with its APK here as well? Thanks!

iGadget, 2022-09-27:
Interesting... Apparently, I have the latest version from the Play Store installed, but in the app it still says v.2.2.2:00?

IzzySoft, 2022-10-21:
[image]

([source](https://www.appbrain.com/app/com.evnotify.app)) Was that update withdrawn then? PlayStore itself also says:

[image]

([source](https://play.google.com/store/apps/details?id=com.evnotify.app&hl=en)) Which means you seem not to have the latest version from there, @iGadget :wink: And here at Github releases, the **3** latest versions are missing. Will you add them, @GPlay97?

GPlay97, 2022-10-22:
It looks like, I accidentally forgot to update the version number within the app settings page, but it is the latest version, as mentioned.

I just added the latest release now. Since the other ones were mostly patches, I will not add them here again, as the new v2 release will probably be the last v2 release. The upcoming new v3 release is waiting soon for everyone!

closed 2022-10-22 by GPlay97

IzzySoft, 2022-10-22:
Thanks a lot! My updater already found it, so the latest version is [available in my repo](https://apt.izzysoft.de/fdroid/index/apk/com.evnotify.app) again. If you grant me another wish, it would be a version without Firebase Analytics so the `Tracking` anti-feature could be dropped :blush: