EVNotify/EVNotify #290: VW-Group nach Update auf ME 3.0 falsche Ladewerte

damianarnold1993, 2022-10-13:
hallo zusammen

ich habe vor einem Monat ein neuer Enyaq Sportline (Software Stand ME3.0) erhalten. Vorher hatte ich ein "normalen" Enyaq mit Software Stand 2.4)

Gleicher OBD Dongle, gleiches Handy für die Übertragung, jedoch andere Ladewerte (bsp. 1072860000 wird bei kW angezeigt).  SoC, Batterietemperatur ist korrekt!

Ich vermute ein Kommafehler oder dass das der Enyaq nich kWh sondern bsp. Ws ausgibt, seit dem Update. Haben das andere Nutzer auch? ID.3, ID.4, ID.5 bekommen ja auch dieselbe Software.. :)

GPlay97, 2022-10-19:
Interessant. Bis sich jemand meldet, wäre das mal interessant näher zu untersuchen. Schreib mir am besten mal eine E-Mail an info@evnotify.de, dann kann man remote vielleicht über den Debug Modus etwas herausfinden.

DerPawel, 2022-12-04:
Was werden für Daten benötigt? Ich fahre auch ein MEB Fahrzeug.

ypoosn, 2022-12-17:
V2.4.0.1 I have got the same problem. Skoda Enyaq iv80. Is this the reason why the drives logging also does not work?

Let me know how and which further debugging support you need.

[image]

DerPawel, 2022-12-17:
The data for drive logging is not provided by the vehicle. From software version 3.0 and higher, only start + end point and distance travelled (odometer start/end) are released.

ypoosn, 2022-12-17:
> The data for drive logging is not provided by the vehicle. From software version 3.0 and higher, only start + end point and distance travelled (odometer start/end) are released.

Thanks for clarification. Shame on VW. Even my $30 smartwatch provides more useful data that this car.

petosak, 2023-02-10:
Hi, I have ID.4 with 3.0 SW. App is absolutely useless. Bad values and no data in logs.....

GPlay97, 2023-02-11:
The app is not useless. Logs are independent from the car, mostly a client side configuration issue. The new firmware updates from vehicles sometimes change behavior, so it needs to be adapted. I'm addressing these firmware changes, but it's hard to maintain, when they change it. On v3, I'm re-implementing the vehicles and I'm thinking of adding a settings for firmware versions, so that the same car can run different configurations based on installed firmware to address this issue

petosak, 2023-02-13:
How I can help? I want help.

QuasarBorn, 2023-02-28:
I have the same problem as the OP. My figure has 109579000...

Cupra Born running ID software 3.0

Please let me know if you would like any more info, logs etc and I'd be happy to help.

MattiJarvinen, 2023-05-28:
Iternio confirmed the message is now a multiframe response. Formula for parsing is the same as before.

@GPlay97 maybe a new ID.3/4/5/ 3.x+ car selection and same for Enyaq. I'm not sure about Audi Q4 but it's very likely.

https://github.com/iternio/ev-obd-pids/issues/6

Mo-to, 2024-01-16:
> Iternio confirmed the message is now a multiframe response. Formula for parsing is the same as before.
> 
> @GPlay97 maybe a new ID.3/4/5/ 3.x+ car selection and same for Enyaq. I'm not sure about Audi Q4 but it's very likely.
> 
> [iternio/ev-obd-pids#6](https://github.com/iternio/ev-obd-pids/issues/6)

Even if this is a few months ago. I don't understand how the formula can be same.
According to Iternio the formula is the following:
```"equation": "((A<<32+B<<16+C<<8+D-150000)/100)*-1",```
https://github.com/iternio/ev-obd-pids/blob/0f3c69fde28c3ae720f9f8b2f5c65f2f1dd2b25e/volkswagen/MEB.json#L26`

Meaning that i basically need 4 data bytes A,B,C,D, which easily fits into the 8 bytes of a normal can frame. Why do they changed it to a multiframe and what additional information can i find in the multiframe?

ptbil94, 2024-05-08:
Hello,

is there an update on this matter? Can't find out how to convert the current...

MattiJarvinen, 2024-05-24:
Iternio PID issue linked above has some updates.