EVNotify/EVNotify #292: notification of new update does not go away after updating to latest APK (2.3.0)

mhantsch, 2022-10-23:
### New issue
- Short description: notification of new update does not go away after updating to latest APK (2.3.0) 
- Detailed description:
On startup, EVNotify shows a notification "Neues Update". I have downloaded the latest APK for 2.3.0 and installed it, but the notification still appears. When I look at the version number at the bottom of the settings page, it still reads "v.2.2.2:00".

- Can you reproduce it? (_If so, how?_): yes, just follow the steps above.

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No): No
  
<details>
  <summary>additional information</summary>
  In [203000-1.apk](https://github.com/EVNotify/EVNotify/releases/download/2.3.0/203000-1.apk)/assets/www/client.js I can see `VERSION = '2.2.2:00'`, so maybe that's causing the issue?
</details>

Tobzi1977, 2022-11-10:
Small addition:
Same behaviour after update from v.2.2.0 to v.2.4.0

GPlay97, 2022-11-12:
Should now be fixed when updating to latest patch.

closed 2022-11-12 by GPlay97

Tobzi1977, 2022-11-12:
Works here. Thanks!