EVNotify/EVNotify #293: No connection to OBDLink CX dongle possible

ypoosn, 2022-11-11:
### New issue
- Short description: No connection to OBDLink CX dongle possible
- Detailed description: Connection to "quality" dongle OBDLink CX fails. PopUp shows bluetooth connection fails. Dashboard shows message "Please connect first time (invalid date)" 

Dongle shows car values with  with ABPR, Skoda Enyak 80, Xiaomi Mi9.

- Can you reproduce it? (_If so, how?_): yes

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No): yes
  APP not usable with this dongle
 <details>
  <summary>additional information</summary>
 recorded debug session around Fri Nov 11 17:30 CET 2022
akey is 35fe8e
</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

ypoosn, 2022-11-12:
update to obd firmware 5.7.1 . Still no connection

ypoosn, 2022-11-12:
obd firmware update to 5.7.1 still no connection possible

noradtux, 2022-11-12:
The CX is a BLE dongle. AFAIK evnotify needs a classic Bluetooth serial dongle. @GPlay97 confirm?

GPlay97, 2022-11-12:
In v2 unfortunately only Classic Bluetooth protocol is supported (on Android). BLE support is only supported in iOS and there only specific uuid support. See Wiki for compatibility.

in v3, the upcoming app generation (currently in separate private repository), this dongle and many other BLE dongles will be supported for Android and iOS as well. It is not planned to add more support and new protocol implementation on v2. Therefore I'm closing it for now.

closed 2022-11-12 by GPlay97