EVNotify/EVNotify #296: EVNotify in car uploads only SOC (%) but not current power (kW)

mhantsch, 2023-01-07:
EVNotify on an Android phone in my car shows SOC at well as current power, but after a few seconds shows a "cloud disconnected" icon. Still, my other Android phone sees the SOC reported correctly, but power consumption is reported at 0 kW.

I did not change anything in my setup, my phones, or my car. Was there a change on the EVNotify servers?

FlNDUS, 2023-01-08:
I have a similar problem. EVNotiPi is sending data an I consume it via the Webapp (and Webapi). Since a few days only the SoC ist updating Power as well as some other metrics is stuck. It won't change, no matter if I drive or charge.
Car is a hyundai ioniq electric 28kWh

GPlay97, 2023-01-08:
Indeed something is wrong for some accounts suddenly. I will inspect it now, what is going on.

GPlay97, 2023-02-13:
This was a temporary backend issue and fortunately is gone