EVNotify/EVNotify #297: No logs registered in EVNotify

pmiddeld, 2023-01-10:
Since January 5, I get no logs of my trips or charges registered. Also, in the charts of the existing logs no data (lines) are shown.

pmiddeld, 2023-01-16:
Update: Charging logs are registered, Driving logs are not

bludevilx, 2023-01-18:
Same issue on my end.

niaendk, 2023-01-18:
I have same issue.

niaendk, 2023-01-27:
I tried creating a new account. But problem remains. No logs at all. 
Really frustrating. 
Might look into ev watchdog instead.

GPlay97, 2023-01-27:
Sorry for responding now. 

There was a temporary issue for a few days in the past, but the issue has been resolved. I can see that logs are registered. But maybe there is another issue on some accounts still? Can you please write me a mail with your akey and token so I can have a look into it? Please write it not here. 

For driving logs to be registered, location permission needs to be allowed (just to ensure. High precision mode)

mhantsch, 2023-01-30:
No, driving logs are still not registered. Only charging. I did not change a single thing on my EVNotify sending device.

pmiddeld, 2023-01-30:
Hi Giuseppe,
Just to let you know that my driving logs are not registered by EVNotify.
My AKey: e0ae0b

Hope you can do something about it.

Best regards,
Peter Middeldorp

pmiddeld, 2023-01-30:
Sorry, forgot the token:
AKey: e0ae0b
Token: dd28c1cb6092c7dbaf89

GPlay97, 2023-01-30:
I will have a deeper Look into it. doesn't seem to be a general issue, since logs are created on the backend. 

I also recorded a few drives today and yesterday. 

Maybe you guys can write your iOS / android version too here in the meantime. Probably there is something on the client side

GPlay97, 2023-01-30:
[image]

This is how it looks on my account

iGadget, 2023-01-30:
> Hi Giuseppe, Just to let you know that my driving logs are not registered by EVNotify. My AKey: e0ae0b Hope you can do something about it. Best regards, Peter Middeldorp Op vr 27 jan. 2023 07:59 schreef GPlay97 ***@***.***>:
> […](#)

@pmiddeld you should probably get a new API key now, because everybody subscribed to this thread got yours in their mail.

iGadget, 2023-01-30:
> I will have a deeper Look into it. doesn't seem to be a general issue, since logs are created on the backend.
> 
> I also recorded a few drives today and yesterday.
> 
> Maybe you guys can write your iOS / android version too here in the meantime. Probably there is something on the client side

Same issue here - I've never been able to get any kind of logging working. Not for charging sessions, not for trips, nothing.
Running EVnotify Android v.2.4.0:01.
Would like to send you my akey and token privately, but I can't seem to find your contact details. Please advise?

GPlay97, 2023-01-30:
> > […](#)
> 
> @pmiddeld you should probably get a new API key now, because everybody subscribed to this thread got yours in their mail.

Absolutely right.

Token should never be shared, as written here and in the issue template. I already invalidated your API key @pmiddeld . You might be able to log into your account (if so, please write a mail to info@evnotify.de)

@iGadget Ok really strange. There are a few conditions when logs are created, but over time logs should appear (especially on longer trips - very short trips will be skipped from the server).
Write me a mail to info@evnotify.de and I can look into it. (Refer to this issue please too).

It is also good to know which OS version you use, for Example Android 9 or so.

iGadget, 2023-01-30:
> Ok really strange. There are a few conditions when logs are created, but over time logs should appear (especially on longer trips - very short trips will be skipped from the server). 

I've been using EVnotify for months now. Short trips, long (150km+) trips... No logs.

> Write me a mail to [info@evnotify.de](mailto:info@evnotify.de) and I can look into it. (Refer to this issue please too).

Done!

> It is also good to know which OS version you use, for Example Android 9 or so.

Android 11 on my main phone, Android 9 on the phone which is usually in the car.

GPlay97, 2023-01-30:
I will answer mails in a few days (I got a lot in the past time).

One thing that also comes in my mind and is sometimes a problem (depending on the phone settings).

The sender phones needs to have location permission allowed for EVNotify and need fine location / high precision mode (so the gps speed can be extracted, which is required (in v2 at least) to create driving logs. Charge logs are created even without location information. Speed is a parameter (extracted from GPS currently only) to ensure that only logs are created, when car really drives and not only standing still.

mhantsch, 2023-01-30:
I don't think it has to do with the location permissions because of two facts:

1. I have an old phone in the car, which is always there, it doesn't receive any updates automatically, and it runs a couple of automation scripts (which I wrote myself using "Automate"). It basically works like a "EVNotiPi". That phone has been unchanged (no OS updates, no EVNotify updates, no change to settings...). From one day to the next, all of a sudden I was not seeing driving logs anymore.

2. My EVNotify is linked to ABRP. I see driving logs in ABRP. I can see all my routes there. So EVNotify (from that fixed phone in my car) is definitely receiving GPS coordinates, and is passing all that information to the ABRP server. But still, I do not see any drives when I check EVNotify (neither using the EVNotify Android app nor via the Web interface). 

Because of these 2 facts, my guess is that there must be something wrong on the EVNotify server.

I'm happy to provide you my akey and token, but where do I need to send it to?

mhantsch, 2023-01-30:
I don't think it has to do with the location permissions because of two facts:

1. I have an old phone in the car, which is always there, it doesn't receive any updates automatically, and it runs a couple of automation scripts (which I wrote myself using "Automate"). It basically works like a "EVNotiPi". That phone has been unchanged (no OS updates, no EVNotify updates, no change to settings...). From one day to the next, all of a sudden I was not seeing driving logs anymore.

2. My EVNotify is linked to ABRP. I see driving logs in ABRP. I can see all my routes there. So EVNotify (from that fixed phone in my car) is definitely receiving GPS coordinates, and is passing all that information to the ABRP server. But still, I do not see any drives when I check EVNotify (neither using the EVNotify Android app nor via the Web interface). 

Because of these 2 facts, my guess is that there must be something wrong on the EVNotify server.

I'm happy to provide you my akey and token, but where do I need to send it to?

[image]

iGadget, 2023-01-30:
> I'm happy to provide you my akey and token, but where do I need to send it to?

info@evnotify.de

pmiddeld, 2023-01-31:
It's Android version 12 for me.

GPlay97, 2023-02-13:
Can you try again? I just had this issue now on my own and I didn't change anything. I just discovered, that the database was under heavy load. I did some maintenance and now logs are working again for me. Maybe this was an issue for you too? Still strange, that for most users it was working, including me. I will have a more detailed look at it.

mhantsch, 2023-02-13:
@GPlay97 I can test from Thursday. I am not with my car right now.

iGadget, 2023-02-14:
> Can you try again?

I charged my Ioniq today with EVnotify running - still no luck/log.

GPlay97, 2023-02-14:
Stopped working again for me too now.

I will try to implement now something on the backend, that slows down the records at all, so server can process the incoming data better.

I will get back to you

pmiddeld, 2023-03-06:
Anything new on this issue? I still get the charging logs, but it's flattened by very short charging power peaks of thousands of kW, increasing the range of the y-axis but that might be a specific Kona flaw.

iGadget, 2023-03-16:
@GPlay97 I guess this issue is more complex than you anticipated? Is there anything I / we can do to help?

je-bor, 2023-03-17:
hallo all zusammen,
wie ich hier lese bin ich nicht der einzige mit einem problemchen hier.
also bei mir geht es gar nicht mehr weder wird die ladung noch gps daten mehr übertragen.
das web interf. zeigt mir dreierlei daten in der vergangenheit an und einen verbrauch wenn das auto aus ist.
und so sieht es aus:
[EVNotify Webinterface.pdf](https://github.com/EVNotify/EVNotify/files/11002477/EVNotify.Webinterface.pdf)

wie ist es bei euch? oder geht wenigstens das laden?
gruß jens

GPlay97, 2023-03-19:
Hey everyone. 

I tried some optimizations tasks on the database and also experimented with a few log creation changes. Not all of them were successful, some made it even worse for most users. I'm trying now one new thing that I will schedule tomorrow evening that hopefully brings some results. If I fail again I might create some topic on it the next days, maybe community can help then. 

I moved to a new apartment and had a lot of stress , that's why I haven't answered earlier. Sorry for that

pmiddeld, 2023-03-20:
Hi Giuseppe,

I had some additional observations which may be relevant (or not):
- For me, charging logs are created, driving logs aren't
- Live data is sent to my receiver phone, both when charging and driving
- Deleting a log (both charging and driving logs) in the app gives an error
message: "Log entry could not be deleted". When going back to the list of
logs, despite the error message, the log is deleted

At least for me to log my drives is "nice to have", so no big hurry as far
as I am concerned.

niaendk, 2023-03-30:
Just bought an used Evnotify PNP unit, at it also have problems with the logging. :/
stops logging while driving and can't se the log history.

iGadget, 2023-04-11:
> I'm trying now one new thing that I will schedule tomorrow evening that hopefully brings some results. If I fail again I might create some topic on it the next days, maybe community can help then.

Please do (if you haven't done so already). Hopefully we can get to the bottom of this together.
 
> I moved to a new apartment and had a lot of stress , that's why I haven't answered earlier. Sorry for that

Oh boy, removals can cause a lot of stress indeed. Hope you've settled a bit in the mean time? At any rate - Take care & don't rush things, nobody wins if you end up in a burn-out (and I know from personal experience that risk can be all too real). Even with all of us coming to 'depend' on your little gem, it's not worth it for you (or anybody else) to get into health issues.
Do ask for help if you want / need it!

MattiJarvinen, 2023-05-09:
Same here. Last log entry date 6.1.2023.

niaendk, 2023-05-27:
Still no logs :/
Any news on the issue?

DakotaBcn, 2023-07-10:
from 5 Jan i no have any log in evnotify