EVNotify/EVNotify #303: evnotify no integrate with ABRP

DakotaBcn, 2023-07-10:
### New issue
- Short description:  ABRP integration fail

- Detailed description:
Is impossible integrate wih evnotify, after push GRANT and OK in ABRP for my Kia Eniro show json message 
<html><body>

  | "Successfully connected. That's it. Your data will now be transfered automatically."
-- | --

</body>
</html>

- Can you reproduce it? (_If so, how?_):

Yes, i probe with S10+, Carlinkit, another telephon and PC, same error

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No): No
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

djoenez, 2023-09-13:
Hello, I tried a couple of times linking Abrp. I get the message that it is successful. But ABRP is not getting data (checked with their support). Also tried reinstalling the app and a different account. Unfortunately it did not worm

miluoshi, 2023-11-02:
I have exactly the same problem and I've also tried creating new accounts.

EVNotify v2.4.0
ABRP v4.6.5
OBD dongle: Vgate iCar Pro BLE 4.0
Car model: Hyundai IONIQ Electric 28 kWh

joopdo, 2024-07-08:
it doesn't work for me either, when I tap integrate with ABRP -> next, I get "Application Error" net::ERR_FILE_NOT_FOUND (file:///android_asset/www/[object%20Window])

joopdo, 2024-07-10:
When I do a logcat I see:
`07-10 19:38:35.007 11147 11147 D SystemWebChromeClient: file:///android_asset/www/cordova.js: Line 315 : Uncaught TypeError: Cannot read properties of undefined (reading 'setMessage')`

GPlay97, 2024-07-12:
Can you try it within web.evnotify.de. It should work there (as a workaround for now)

joopdo, 2024-07-13:
wow! i didn't realise there exists a web version. this seems to work, i don't see any error any more! thank you!

Micr0mega, 2024-11-20:
I have a similar problem linking to ABRP, but I do get an oauth error from ABRP:

[image]

> Can you try it within web.evnotify.de. It should work there (as a workaround for now)

I have used the web interface as well as the app, both give me the same oauth error.

danielStrobl, 2025-02-07:
I have the same oauth error, both via the app and the web version.