EVNotify/EVNotify #313: API not accepting data for me from 2 phones using separate aKeys and different data connections as of 17 hours ago

caitotheonlypotato, 2024-03-06:
### New issue
- API not receiving updates: 
- Phone in both cars try to upload and then the uplod icon changes to a cloud with a line through it.  When querying the website directly only old data is available (17 hours ago at the time of submitting this.  Going to the API URL gives a certificate error, not sure if that's related.:

- Can you reproduce it? (_If so, how?_)
- Yes on 2 phones with separate data connections
- :

- [ ] critical issue? (_issue, which will affect multiple accounts, 
yes, affects 2 saparate aKeys
  
 <details>
  <summary>Can download last updated data but can't upload new data</summary>
</details>

djungelola, 2024-03-06:
Have the same problem. Think it is a certificate issue.

GPlay97, 2024-03-06:
Can you share more details on what url you get the error?

mhantsch, 2024-03-06:
Seems to still work for me...

GPlay97, 2024-03-06:
For me it seems to work too. But indeed yesterday the certificates switched, therefore I thought they might still be some issue.

If you have an certificate issue, please post the full error response along with device information.

GET https://app.evnotify.de/key gives me a response.

Keep in mind , that api.evnotify.de points to the (still unused) v3 api (this url currently has no active certificate), but does not influence the "live" v2 api

caitotheonlypotato, 2024-03-06:
[image]
[image]
[image]

Also, on the phone, I tried logging out, but I can't log back in.  I also can't create a new account

caitotheonlypotato, 2024-03-06:
Login error
[image]
Create new account error
[image]

Was woring and then stopped without any modifications made clientside

GPlay97, 2024-03-06:
As written earlier, the domain "api.evnotify.de" is the upcoming v3 API , which is not public available yet and NOT influences current app.

The domain "app.evnotify.de" is used for v2 app and it is valid.

This can also be seen within the file @djungelola sent

[image]

I checked the "old" v2 code now locally and built for browser and works fine.

Looks to me like thats an android issue now with the old outdated code.. 

I'll have a deeper look

GPlay97, 2024-03-06:
@caitotheonlypotato Can you try to completely uninstall the app and retry?

I installed now a fresh installation on my real phone and it works.

I can also see, that users are using the app normally.

This does not mean, that there is no issue at all, still can be an issue, but first, try to uninstall, maybe something was cached with the certificate that expired yesterday?

GPlay97, 2024-03-06:
I See successful login attempts on the backend and since I enabled now monitoring one "bad" login, but from the error it says that it is wrong credentials.

If some of you still have issues, please write here the AKey or via mail info(at)evnotify.de - I'll have a look at the account, I can also set the password to a random one.

At least on the backend and also on the phone I used for testing now, I can't see a general error yet. So anyone still having issues, please write here or via mail with more details, so I can troubleshoot

djungelola, 2024-03-06:
Hi
I still have the problem. Tested to delete and re-install my app.
My aKey=08f24e
Just tested to log in, get the same error as you can see in the screenshots posted above.
Let me know if i can do anything else to help you troubleshoot.

/Ola

caitotheonlypotato, 2024-03-06:
my aKey is 88c183

GPlay97, 2024-03-06:
Interesting, can't see any logs of both of your AKeys.

It looks like it is blocked now on client side, but why? I'm investigating and trying to reproduce this on a phone, or maybe an emulator. Might be a specific android version now? Or try to look for Android System Webview in Play Store. Looks like there was an update, maybe this caused it? The current v2 version uses a webview

GPlay97, 2024-03-06:
Looks like I was right with WebView.

Just found some article and a post with a similiar issue: https://forum.bubble.io/t/all-my-applications-loses-access-to-db-under-a-webview/310975

caitotheonlypotato, 2024-03-07:
thought I'd have a try of a few things.  I disabled webview (also goes back to original factory version).  Could uninstall and reinstall EVNotify, but can't authenticate still.  I then updated to the latest Android SYstem Webview ( for me on Android 5 is 96.0.4664.27), same problems.  I uninstalled again and then installed v2.3 apk from github same problems.

If there's somethng else I can try, I'm happy to give it a go.

GPlay97, 2024-03-07:
Do you have google services on the phone? At least on the one thread listed, the issue was on phones, that dont have Google Mobile Services (GMS) . Would  be a hint, and an explanation, why it works on my phones, where I tried

djungelola, 2024-03-07:
I have GMS on my phone.
Will try to find another android with newer version.

djungelola, 2024-03-07:
can this be related?
https://medium.com/@anthony_chaffey/ssl-certificates-no-longer-supporting-android-7-0-335677be17ec

mhantsch, 2024-03-07:
Throw some LineageOS on the older phones? I also have an older OnePlus One in the car with LineageOS 18.1 (Android 11) and that one works without issues.

caitotheonlypotato, 2024-03-07:
> Throw some LineageOS on the older phones? I also have an older OnePlus One in the car with LineageOS 18.1 (Android 11) and that one works without issues.
For sure, I did start looking, the ASUS Z00AD effectively stops at Lin OS14 (the 15 is super buggy and a lot of things don't work),.  That's where the Z00AD seems to end.  The HTC one M9 is a possibilty

caitotheonlypotato, 2024-03-08:
> Throw some LineageOS on the older phones? I also have an older OnePlus One in the car with LineageOS 18.1 (Android 11) and that one works without issues.

I managed to get LineageOS 18.1 onto  HTC One M9 and root it (with some resistance from the phone I might add).  Now all is back to normal on the updated OS

EggieJ, 2024-03-15:
I have the same problem. 
Trying to set up EVNotify on an LG K4 LTE with Android 5.
Can't login to my existing account with aKey 000fee, neither can I create a new account from this phone.

Tried re-installing the app. 
The phone has original OS, so including GMS (I assume)

mhantsch, 2024-03-15:
> I have the same problem

EVNotify cannot fix this. The problem lies in the underlying Android. You have to use a newer Android version (I think at least Android 10 is needed).

What would you hope to achieve by adding another comment here?

I believe this issue can be closed.

EggieJ, 2024-03-15:
> EVNotify cannot fix this. The problem lies in the underlying Android. You have to use a newer Android version

Alright, I didn't see that was the (only) conclusion.

GPlay97, 2024-04-06:
https://letsencrypt.org/certs/isrgrootx1.pem
https://letsencrypt.org/certs/lets-encrypt-r3.pem

You can install these two certificates manually on the old android phones, then it will work again :)

closed 2024-04-06 by GPlay97