EVNotify/EVNotify #314: Ist das Projekt tot??? Is the project dead???

jwinkler1967, 2024-03-06:
Hello everyone, does this project still exist???

GPlay97, 2024-03-06:
No the project is not dead.

I had (and still not finished) several health issues in family, which was the main reason why I was not active for a long time.

Even before that, I was not super active unfortunately, due to many support inquiries, PlugAndPlay devices etc. But I discontinued (will not sell new ones), and therefore work on v3 software only again.

There will be an early access version of the v3 version, search on Telegram for the evnotify channel, there it will be announced first.

jwinkler1967, 2024-03-06:
Thank you for your feedback, it's great to hear that you're feeling better! I'm happy to hear that the project is continuing. I came to EN Notify via "EVCC" and find the whole networking of the topics of PV, EV and SmartGrid very exciting. I'm not that active on Telegram, but I'll take a look at it now. Thank you again for the feedback.
The app in the EV hasn't been running since yesterday evening, is there a central reason for this? or do I have to keep looking?!

GPlay97, 2024-03-06:
I'm looking into it, looks related to that: https://github.com/EVNotify/EVNotify/issues/313