EVNotify/EVNotify #318: App ist aus dem AppStore verschwunden

walterleonhard, 2025-01-06:
Neues iPhone gekauft, jetzt ist die EV-Notify App weg und kann auch nicht mehr installiert werden, weil es sie im AppStore nicht mehr gibt. Auf meinem alten Android-Gerät funktioniert das Einloggen nicht, auch eine neue Account-Erstellung geht nicht, Android-Version anscheinend zu alt.
Naja, jetzt stehe ich im Regen mit diesem Gerät.

Die auf der Homepage angegebene e-Mailadresse funktioniert nicht, also muss ich mich hier an euch wenden.

mhantsch, 2025-01-06:
Zuletzt war der Entwickler auf Telegram Channel "EVNotify" aktiv. Ev. dort auch noch probieren...

GPlay97, 2025-01-06:
Hallo @walterleonhard !

Ja, das ist keine gute Situation, ich weiß. Bei Apple muss man jährlich die Lizenz neu kaufen für 100$, die aber nur per Kreditkarte zu zahlen ist. Da ich keine habe, habe ich eine virtuelle damals genutzt, das ging problemlos, bis zur letzten Verlängerung. Scheinbar sperrt Apple manche Kreditkarten Anbieter inzwischen.

In Zukunft wird es aber definitiv wieder eine iOS Version geben - aller Voraussicht nach aber erst mit der neuen EVNotify App "v3".

Unter Android ist das Ganze einfacher. Der Grund, wieso das bei deinem ganz alten Gerät aber nicht mehr geht, ist, dass die Zertifikate nicht mehr akzeptiert werden (s. hier: https://medium.com/@anthony_chaffey/ssl-certificates-no-longer-supporting-android-7-0-335677be17ec). Um dem entgegen zu wirken, kann man die Zertifikate aber auch manuell installieren (s. hier: https://github.com/EVNotify/EVNotify/issues/313#issuecomment-2041025514).

Alles nicht schön, ich weiß. Ich arbeite daher jetzt besonders auf die V3 hin - die Alpha Phase ist da gestartet und wird - zunächst über Firebase App Distribution - als APKs verteilt.

iGadget, 2025-01-07:
@GPlay97 good to hear from you, hope you are doing well. Best wishes for 2025!
If there's anything I  /  we can do to help you test V3, please let us know!

walterleonhard, 2025-01-07:
Danke, gut zu wissen, dass es weiter geht.
Wenn es für iOS etwas zum Testen gibt, bitte gerne.